embassy-rs/nrf-softdevice #2: Macro error

kuon, 2020-09-12:
I'm trying to dive in and try your set of crate, but I have an error that maybe you have an idea where it originates from:

```
error[E0425]: cannot find value `irq` in this scope
   --> /home/kuon/.cargo/git/checkouts/nrf-softdevice-9b99539d60cc72a7/52329d2/nrf-softdevice/src/util/macros.rs:30:40
    |
30  |                 defmt::error!($msg, $( $i ),*);
    |                                        ^^ not found in this scope
    |
   ::: /home/kuon/.cargo/git/checkouts/nrf-softdevice-9b99539d60cc72a7/52329d2/nrf-softdevice/src/interrupt.rs:248:5
    |
248 |     assert_app_accessible_irq!(irq);
    |     -------------------------------- in this macro invocation
    |
```

kuon, 2020-09-13:
I published a playground repository I will use to test this, I have the error with it currently:

https://github.com/kuon/nrf-playground

Dirbaio, 2020-09-13:
Thanks for reaching out! 

That macro builds fine for me, I'm using `rustc 1.48.0-nightly (dbb73f8f7 2020-09-12)`. What rust version are you using? You should use a recent nightly.

By the way, I've just pushed some changes to make nrf-softdevice build properly with the last released cortex-m (instead of git master). You should remove the `nrf52840-pac` and `cortex-m` patches from Cargo.toml and run `cargo update`.

kuon, 2020-09-13:
Yeah, I was on 1.47 and after upgrading to 1.48 it built.

closed 2020-09-13 by kuon

kuon, 2020-09-14:
Perhaps add a line at the top of the readme specifying the 1.48 as minimum requirement? Could  save some other folks time.