embassy-rs/nrf-softdevice #10: GATT server

Dirbaio, 2020-09-13:
This'll be a fun one

kuon, 2020-09-15:
I guess you are still working on it.

For my current project, I need the to provide custom data to my bluetooth client app (running on my phone).

I put the basics here:

https://github.com/kuon/nrf-playground

Now, I generate 4 random bytes instead of the read data packet. I put together the general structure to be able to to `gatt_server::set_value` from my timer task.

I am trying to connect to the nrf52840 from the nordic "nrf connect" bluetooth Android app. I can see it, but when I do connect, the app just timeout.

I tried your examples, and they produce the same result, so I guess this is still a work in progress.

Can I help with something?

Dirbaio, 2020-09-16:
It's probably timing out due to #5 or #6 not being implemented. My phone (Pixel 3A) times out due to #5. See if you're getting an `on_data_length_update_request` or `on_exchange_mtu_request` logged when you connect from the phone. If you do, that's the issue, the phone is waiting forever for a reply that we never send.

(update to latest master, I just added the `on_data_length_update_request` log)

kuon, 2020-09-16:
Yeah, I got:

```
0.000017 TRACE on_data_length_update_request conn_handle=3
└─ nrf_softdevice::ble::events::on_data_length_update_request @ /home/kuon/.cargo/git/checkouts/nrf-softdevice-9b99539d60cc72a7/36b0d99/nrf-softdevice/src/ble/events.rs:256
0.000018 TRACE on_sec_params_request conn_handle=3
└─ nrf_softdevice::ble::events::on_sec_params_request @ /home/kuon/.cargo/git/checkouts/nrf-softdevice-9b99539d60cc72a7/36b0d99/nrf-softdevice/src/ble/events.rs:166
0.000019 TRACE gatts on_exchange_mtu_request conn_handle=3
└─ nrf_softdevice::ble::gatt_server::on_exchange_mtu_request @ /home/kuon/.cargo/git/checkouts/nrf-softdevice-9b99539d60cc72a7/36b0d99/nrf-softdevice/src/ble/gatt_server.rs:246
0.000020 TRACE on_disconnected conn_handle=3
└─ nrf_softdevice::ble::events::on_disconnected @ /home/kuon/.cargo/git/checkouts/nrf-softdevice-9b99539d60cc72a7/36b0d99/nrf-softdevice/src/ble/events.rs:146
0.000021 TRACE conn 0: disconnected
└─ nrf_softdevice::ble::connection::{{impl}}::on_disconnected @ /home/kuon/.cargo/git/checkouts/nrf-softdevice-9b99539d60cc72a7/36b0d99/nrf-softdevice/src/ble/connection.rs:108
```

Dirbaio, 2020-11-09:
Closing, this is mostly done. WIll open individual issues for missing things.

closed 2020-11-09 by Dirbaio