embassy-rs/nrf-softdevice #14: Investigate PPI safety

Dirbaio, 2020-09-13:
The softdevice has a bunch of `sd_ppi_*` functions, which makes me think we can't freely use PPI when softdevice is enabled. This may conflict with PPI usage in nrf-hal or other libs.

At the very least we should document it.

jacobrosenthal, 2020-09-15:
It looks like we might be just restricted from using channels 17-31 and groups 4 and 5 
We can consume the appropriate channels easy enough.. Im not sure what to do about groups 4 and 5
https://github.com/nrf-rs/nrf-hal/pull/212 Thoughts?

----

From https://devzone.nordicsemi.com/f/nordic-q-a/45337/ppi-resources-used-by-s140
>  The PPI peripheral can be configured directly in the application. In the very first versions of the SoftDevice (for nRF51 series), there was an API for using the PPI, but now you can access it directly on a register level.

> If you try to access (ie: write) to a peripheral which the softdevice controls, like the radio, it will cause a fault condition.

I guess theyre in the headers if we want to generate them, but for the headers we have theyre all the same

link For s140
https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v7.0.1/group___n_r_f___s_o_c___d_e_f_i_n_e_s.html

Mask of PPI channels reserved by the SoftDevice when the SoftDevice is enabled. 
#define NRF_SOC_SD_PPI_CHANNELS_SD_ENABLED_MSK
((uint32_t)( \
      (1U << 17) \
    | (1U << 18) \
    | (1U << 19) \
    | (1U << 20) \
    | (1U << 21) \
    | (1U << 22) \
    | (1U << 23) \
    | (1U << 24) \
    | (1U << 25) \
    | (1U << 26) \
    | (1U << 27) \
    | (1U << 28) \
    | (1U << 29) \
    | (1U << 30) \
    | (1U << 31) \
  ))

Mask of PPI groups reserved by the SoftDevice when the SoftDevice is enabled. 
#define NRF_SOC_SD_PPI_GROUPS_SD_ENABLED_MSK
((uint32_t)( \
      (1U << 4) \
    | (1U << 5) \
  ))

jacobrosenthal, 2020-09-15:
Older link but
https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s140.api.v6.1.0%2Fgroup___n_r_f___s_o_c___e_n_u_m_s.html

It looks like the api calls are for when softdevice is disabled, to be able to work with those restricted channels and groups
SD_PPI_CHANNEL_ENABLE_GET | Base value for SVCs that are available when the SoftDevice is disabled.
-- | --
SD_PPI_CHANNEL_ENABLE_SET | Base value for SVCs that are available when the SoftDevice is disabled.
SD_PPI_CHANNEL_ENABLE_CLR | Base value for SVCs that are available when the SoftDevice is disabled.
SD_PPI_CHANNEL_ASSIGN | Base value for SVCs that are available when the SoftDevice is disabled.
SD_PPI_GROUP_TASK_ENABLE | Base value for SVCs that are available when the SoftDevice is disabled.
SD_PPI_GROUP_TASK_DISABLE | Base value for SVCs that are available when the SoftDevice is disabled.
SD_PPI_GROUP_ASSIGN | Base value for SVCs that are available when the SoftDevice is disabled.
SD_PPI_GROUP_GET | Base value for SVCs that are available when the SoftDevice is disabled.

Dirbaio, 2022-02-05:
In the end enforcing this kind of safety at compile time needs coupling with a particular HAL (embassy-nrf or nrf-hal) to consume their singletons. What's worse, needs coupling to a *particular major version* of the HAL.

Due to that, we're already not consuming singletons for "whole" softdevice-reserved peripherals, so it's unlikely we'll do that for individual PPI channels/groups either.

closed 2022-02-05 by Dirbaio