embassy-rs/nrf-softdevice #17: Disallow declaring IRQ handlers for reserved interrupts

Dirbaio, 2020-09-14:
Currently this builds fine in user code

```rust

#[interrupt]
fn RADIO() {
    info!("RADIO triggered!")
}

```

but will not work since RADIO is a softdevice-reserved interrupt. I *think* it does work before enabling the softdevice, and after enabling it'll just never fire (the softdevice never forwards that irq to the application).

Not sure if this should be somehow prevented from compiling (if that's even possible).

jacobrosenthal, 2020-09-15:
As long as SD will steal them, Lets just define them ourselves. Then theyll be overriden, which is fine.

If someone tries to use it itll fail for being defined twice
https://github.com/rust-embedded/cortex-m-rt/blob/6c4689de32546ee1d271270a1e6e5a0fe52266de/tests/compile-fail/interrupt-twice.rs

```
  = note: rust-lld: error: duplicate symbol: RADIO
          >>> defined at rtic.rs:143 (examples/src/bin/rtic.rs:143)
          >>>            /home/jacob/Downloads/nrf-softdevice/target/thumbv7em-none-eabihf/debug/deps/rtic-6ac6ec1c4166c100.rtic.bl7ebdu1-cgu.0.rcgu.o:(RADIO)
          >>> defined at lib.rs:47 (nrf-softdevice/src/lib.rs:47)
          >>>            nrf_softdevice-2b21128dbb5f1edc.nrf_softdevice.besa5utw-cgu.0.rcgu.o:(.text.RADIO+0x1) in archive /home/jacob/Downloads/nrf-softdevice/target/thumbv7em-none-eabihf/debug/deps/libnrf_softdevice-2b21128dbb5f1edc.rlib
```

Dirbaio, 2020-11-09:
In the end I think it's better to not do this, because you CAN use reserved interrupts before enabling the softdevice, or if you never enable the softdevice at all (some device with a physical switch that switches it between BLE and custom radio protocols, for example)

closed 2020-11-09 by Dirbaio