embassy-rs/nrf-softdevice #19: Enable print-defmt feature for panic prob in examples

kuon, 2020-09-14:
The examples `Cargo.toml` could have `panic-probe` enabled with the `print-defmt` feature:

`panic-probe = {version = "0.1.0", features = ["print-defmt"]}`

Dirbaio, 2020-09-14:
The issue with `print-defmt` is that it wastes flash space because the panic message strings are not replaced with IDs like normal defmt prints, they're stored entirely in the flash. It also pulls in `core::fmt` to format the panic message, which is a bit heavy.

Instead, both the examples and the `nrf-softdevice` lib use a `depanic!` macro that prints with defmt and then panics. Together with utilities like `dewrap`, `dexpect`, `deassert!`. This way we get panic messages without the flash space hit.

closed 2020-09-14 by Dirbaio

kuon, 2020-09-14:
I see, this makes sense.