embassy-rs/nrf-softdevice #23: Upgrade to sdk 17.0.2

jacobrosenthal, 2020-09-15:
No hurry, I more wanted to document that the current softdevice thats checked in (do you want to check that in?) and the generated files are from 16.0.0

Dirbaio, 2020-09-15:
A related unsolved question is how does the `nrf-softdevice-s1xx` crate versions relate to the softdevice versions...

jacobrosenthal, 2020-09-15:
16.0.0
https://infocenter.nordicsemi.com/topic/sdk_nrf5_v16.0.0/index.html

    S112 SoftDevice v7.0.1 (API documentation)
    S113 SoftDevice v7.0.1 (API documentation)
    S132 SoftDevice v7.0.1 (API documentation)
    S140 SoftDevice v7.0.1 (API documentation)
    S212 SoftDevice v6.1.1 (API documentation)
    S312 SoftDevice v6.1.1 (API documentation)
    S332 SoftDevice v6.1.1 (API documentation)
    S340 SoftDevice v6.1.1 (API documentation)

17.0.2
https://infocenter.nordicsemi.com/topic/sdk_nrf5_v17.0.2/index.html

    S112 SoftDevice v7.2.0 (API documentation)
    S113 SoftDevice v7.2.0 (API documentation)
    S122 SoftDevice v8.0.0 (API documentation)
    S132 SoftDevice v7.2.0 (API documentation)
    S140 SoftDevice v7.2.0 (API documentation)
    S212 SoftDevice v6.1.1 (API documentation)
    S312 SoftDevice v6.1.1 (API documentation)
    S332 SoftDevice v6.1.1 (API documentation)
    S340 SoftDevice v6.1.1 (API documentation)

jacobrosenthal, 2020-09-15:
Sucks to spread this info across the whole repo.. but should probably document sdk version and sd version in the sd crates

When the time comes, this is the migration guide for 16 ->17
https://infocenter.nordicsemi.com/topic/sdk_nrf5_v17.0.0/migration.html

Looks really minimal for ble/sd changes

Dirbaio, 2020-09-15:
the softdevices currently in the repo are actually from SDK v17.0.0

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_nrf5_v17.0.0%2Findex.html (not linked from the tree? why, Nordic?)

```
    S112 SoftDevice v7.0.1 (API documentation)
    S113 SoftDevice v7.0.1 (API documentation)
    S122 SoftDevice v8.0.0 (API documentation)
    S132 SoftDevice v7.0.1 (API documentation)
    S140 SoftDevice v7.0.1 (API documentation)

```

(quick way to check: the nrf_sdm.h has `SD_*_VERSION` defines)

kuon, 2020-09-15:
Can't we distribute the softdevice in the repo? This would ensure compatibility.

Dirbaio, 2020-09-15:
Yes, they're under the same Nordic license as the headers. Can't see why not.

alexmoon, 2022-09-16:
the softdevices in the repo are from SDK v17.x.x, and due to nordic's deprecation are unlikely to ever be updated. I think we can close this.