embassy-rs/nrf-softdevice #24: nrf-softdevice-gen broken

jacobrosenthal, 2020-09-15:
Fallout from  #1

moving the .cargo/config up means nrf-softdevice-gen is currently broken unless you pass an override --target or just comment out the target in .cargo/config

Mixing std and no_std in a single workspace is kind of a pain

You probably want to move it outside the other no_std stuff or the entire repo?

Dirbaio, 2020-09-15:
True. That's the reason why `.cargo/config` was in `examples/`, but that means you can't build/run examples from the root so it isn't ideal either :( 

Not sure what to do.

Relevant issue: rust-lang/cargo#8112

kuon, 2020-09-15:
One thing I often do is to create a Makefile to wrap language specific tools (rake, webpack, iex, cargo...). Then you just run `make build`, `make gen`... And you ensure correct environment in the tasks. Of course this is just a suggestion.

commit f72fc320 (2020-12-16): Move .cargo/config to examples. Fixes #24
closed 2020-12-16 by Dirbaio