embassy-rs/nrf-softdevice #26: How to have exclusive connection

kuon, 2020-09-16:
How can I do the following?

- start advertizing
- when I got a connection, stop advertizing and don't accept new connection
- restart advertizing when the connection is closed by the the client

Now in my advertize loop, I have the following:

```rust
    loop {
        info!("Advertising start!");
        let conn = peripheral::advertise(
            sd,
            peripheral::ConnectableAdvertisement::ScannableUndirected {
                adv_data,
                scan_data,
            },
        )
        .await
        .dewrap();

        info!("Advertising done!");

        // Detach the connection so it isn't disconnected when dropped.
        conn.detach();
    }

```

I feel like I should do some loop instead of `conn.detach()`  but I don't know if you exposed the APIs yet.

Dirbaio, 2020-09-16:
We'd have to add some sort of `conn.wait_for_disconnected()` function for this I think

Dirbaio, 2020-09-16:
The `gatt_server::run` function can now be used to do this. It runs the GATT server on the connection, and returns when it disconnects.

A similar function `gatt_client::run` coming soon.

Here's the code from ble_bas_peripheral example:

https://github.com/akiles/nrf-softdevice/blob/9aa2ad0377b4b75f7a8602a51bf9d4eea2a36e84/examples/src/bin/ble_bas_peripheral.rs#L99-L105

closed 2020-09-16 by Dirbaio

kuon, 2020-09-16:
Good, I think this is a nice approach.

On a side note, I am wondering if it wouldn't be a good idea to ensure that all softdevice function takes the `Softdevice` argument as first argument, to ensure that the compiler will protect it's ownership.

Dirbaio, 2020-09-16:
Yes, that's the approach I'm following. All functions take either a `&Softdevice` or a `&Connection`.

Functions taking a `&Connection` don't need to take `&Softdevice`, beacuse the only functions constructing a Connection are advertise/connect, that take a `&Sofdevice`. so if you have a Connection instance then the softdevice is guaranteed to be enabled.

kuon, 2020-09-16:
Yeah, makes sense.