embassy-rs/nrf-softdevice #32: Examples do not compile

dthoma, 2020-11-02:
Compilation of the examples fails with:
```
error[E0599]: no method named `dewrap` found for enum `core::result::Result<(), SpawnError>` in the current scope
   --> examples/src/bin/ble_bas_central.rs:163:42
```

```
 note: the method `dewrap` exists but the following trait bounds were not satisfied:
            `SpawnError: Format`
            which is required by `core::result::Result<(), SpawnError>: example_common::Dewrap<()>`
```

I assume that the derived implementations for defmt are not being generated.

Dirbaio, 2020-11-02:
I've recently updated embassy to make defmt optional, and this repo wasn't updated yet. It's updated now, please try again!

I've added the Cargo.lock file to the repo, to make sure it gets built with the right version of the git dependencies.

dthoma, 2020-11-02:
Amazing response time ;-)
Yes everything works fine now, thank you.

closed 2020-11-02 by dthoma