embassy-rs/nrf-softdevice #33: Freeze after Disconnect

dthoma, 2020-11-05:
When I run any of the ble_bas_peripheral example, the device (nrf52840 dk) seems to freeze. It does not go back to advertising nor are there any tracing messages. I also ran the gatt server from within the rtic example and the tick task seems to stop as well.
The disconnect event is not logged before that happens.

It feels like this is actually the softdevice crashing, but maybe it is not configured correctly. Any idea what could cause this or how to track it down?

Dirbaio, 2020-11-05:
I can't reproduce it. What device are you using to connect to the dk? Note that from Android (and probably others) it's still a bit iffy due because we're missing #27 . It works stably when making two nrf's talk to each other (2 custom rust programs, or using the nrf connect for desktop app)

Probe-run outputs a stacktrace on control+c, maybe that can give us a clue of where it hangs? Also check that you're building probe-run and the examples with the same defmt version (same commit).

dthoma, 2020-11-05:
Thanks, it was my fault. I hadn't pulled your very last commit, somehow I managed to pull just before that and thought I was up to date. The issue was due to

```let executor = EXECUTOR.put(Executor::new(cortex_m::asm::wfi));```

instead of sev.

I use whatever my laptop has and gatttool to connect. I got the same issues with Android, I managed to quickfix that however by adding

```
raw::sd_ble_gap_phy_update(gap_evt.conn_handle, &raw::ble_gap_phys_t {
        rx_phys: 0x01,
        tx_phys: 0x01,
    });
```
and
```
raw::sd_ble_gap_data_length_update(gap_evt.conn_handle, core::ptr::null(), core::ptr::null_mut());
```
to `on_phy_update_request` and `on_data_length_update_request`. So I keep the 1mbit phy and use whatever softdevice choses automatically. A proper API is of course desirable, but that works for now. Amazing project btw.

Dirbaio, 2020-11-09:
> factor out the existing one from nrf-softdevice into a separate crate

That won't work, the Uuid type makes softdevice calls. It should be a different one, probably in the macro crate.

The softdevice handles 128bit uuids in a funny manner: you register them at runtime to get an "uuid type", and then to use it you specify the type, and bytes 12-13.  This means if you do a custom service that requires many UUIDs, you can save flash and ram by making them only differ in bytes 12-13: you only register the "base" uuid once, then use the same uuid type with just different bytes 12-13.

If we want to take advantage of this we could make a macro like this:

```rust
uuids!(
    FOO_SVC = "4a04fbec-229b-11eb-98bd-2f03a9d5f786";
    FOO_CHAR_1 = "4a04fbec-229b-11eb-98bd-2f030000f786";
    FOO_CHAR_2 = "4a04fbec-229b-11eb-98bd-2f030001f786";
    FOO_CHAR_3 = "4a04fbec-229b-11eb-98bd-2f030002f786";
    FOO_CHAR_4 = "4a04fbec-229b-11eb-98bd-2f030003f786";
    FOO_CHAR_5 = "4a04fbec-229b-11eb-98bd-2f030004f786";
    FOO_CHAR_6 = "4a04fbec-229b-11eb-98bd-2f030005f786";
)
```

which would generate code to register the "base uuid" just once if uuids are equal in bytes other than 12-13

Not sure if it's worth the effort though :|

Dirbaio, 2020-11-09:
Latest master uses the uuids from the attributes! It's implemented the "lazy" way for now. 

Closing this. I've created #34 to track the more efficient uuid handling, but it's going to be lower priority for now.

closed 2020-11-09 by Dirbaio