embassy-rs/nrf-softdevice #35: Setting gatt value from interrupts

kuon, 2020-11-11:
I see that many things changed in the codebase.

I am also back on my ble project, and I am trying to have a gatt server working.

I am looking at: https://github.com/akiles/nrf-softdevice/blob/master/examples/src/bin/ble_bas_peripheral.rs 

How am I supposed to set the gatt server value from within the code? I mean, for example, imagine I have a timer interrupt in which I read some values from an i2c peripheral, how can I write it to `BatteryService.foo` for currently connected devices to get the update? (I still use rtic, but I don't think it should change much)

Dirbaio, 2020-11-11:
You can set the value with `server.battery_level_set(val)`. You can send a notification to a connection with `server.battery_level_notify(&conn, val)`.

The `Softdevice` and `Connection` objects are not `Send`, so you can't share them between main code and interrupts (you can only use them from a single priority level). You'll have to send the values from the timer interrupt to the main code, using something like an async Channel or Signal, and set them in the server there.

kuon, 2020-11-11:
I see, but when `gatt_server::run` is running, it blocks the soft device thread, so I have no opportunity to do something like reading a channel with `recv()`.

Is it safe to do something like `join!(gatt_server_run_future, my_polling_recv_future)`?

Dirbaio, 2020-11-11:
Yes, that should be fine.

fnafnio, 2021-01-08:
I think my question fits here quite well,
I've run into a similar situation @kuon encountered, where I need to update data in the `Softdevice` from somewhere else in the firmware, preferably without using an allocator.

I'd like to use some kind of mpsc or spsc queue which could be awaited with the connection/server to update the attribute values.

Would something like this (especially the queues) make sense to add to this crate or more likely the embassy crate?

Dirbaio, 2021-01-08:
`futures-intrusive` has async channels that you can just `.await` and work in no-std, no-alloc, check them out :)

closed 2022-02-05 by Dirbaio