embassy-rs/nrf-softdevice #37: Failing to build on OS X

huntc, 2020-12-26:
Very excited to use this project and help where I can!

I'm finding the getting started a bit rough. Out of the box, and noting that I'm using OS X:

```
% ./test-build.sh 
+ cd examples
+ cargo build --target thumbv7em-none-eabihf --features cortex-m-rtic --bins
  Downloaded rtic-syntax v0.4.0
  Downloaded cortex-m-rtic v0.5.5
  Downloaded cortex-m-rtic-macros v0.5.2
  Downloaded rtic-core v0.3.0
  Downloaded 4 crates (149.3 KB) in 1.05s
   Compiling stable_deref_trait v1.2.0
   Compiling vcell v0.1.2
   Compiling bitfield v0.13.2
   Compiling r0 v0.2.2
   Compiling futures-core v0.3.8
   Compiling futures-sink v0.3.8
   Compiling scopeguard v1.1.0
   Compiling autocfg v1.0.1
   Compiling futures-task v0.3.8
   Compiling pin-utils v0.1.0
   Compiling nb v1.0.0
   Compiling futures-io v0.3.8
   Compiling hashbrown v0.9.1
   Compiling void v1.0.2
   Compiling anyfmt v0.1.0 (https://github.com/akiles/embassy#2e062f56)
   Compiling rand_core v0.5.1
error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: unknown debugging option: `trap-unreachable`

error: could not compile `scopeguard`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: build failed
```

I also note that CI is failing presently at commit `41f00393ddd5607c415ecc3d1862703c85411820`. However, previous working CI commit also fails for me. Something to do with my OS X environment?

huntc, 2020-12-26:
Ignore. I'll update the README to ensure that `rustup update` is done first. :-)

closed 2020-12-26 by huntc

Dirbaio, 2020-12-26:
Yeah, the `trap-unreachable` flag is from a very recent nightly!

You can also just remove it from .cargo/config, it's only to get slightly more compact code.