embassy-rs/nrf-softdevice #38: Cannot build the example having built the library

huntc, 2020-12-26:
Having difficulty getting started with an example noting that I've successfully run the `test-build.sh` script.

As per the README, if I try:

```
cargo run --bin ble_bas_peripheral
```

...then:

```
% cargo run --bin ble_bas_peripheral 
   Compiling nrf-softdevice-s140 v0.1.1 (/Users/huntc/Projects/hacking/nrf-softdevice/nrf-softdevice-s140)
   Compiling cortex-m-rt v0.6.13
warning: unused import: `core::str::FromStr`
 --> nrf-softdevice-macro/src/lib.rs:5:5
  |
5 | use core::str::FromStr;
  |     ^^^^^^^^^^^^^^^^^^
  |
  = note: `#[warn(unused_imports)]` on by default

warning: 1 warning emitted

   Compiling nrf52840-pac v0.9.0
   Compiling panic-probe v0.1.0
LLVM ERROR: Global variable 'Reset' has an invalid section specifier '.Reset': mach-o section specifier requires a segment and section separated by a comma.
error: could not compile `cortex-m-rt`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
error: invalid register `r0`: unknown register
    --> nrf-softdevice-s140/src/bindings.rs:1737:9
     |
1737 |         inout("r0") to_asm(p_mutex) => ret,
     |         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

error: invalid register `r1`: unknown register
    --> nrf-softdevice-s140/src/bindings.rs:1738:9
     |
1738 |         lateout("r1") _,
     |         ^^^^^^^^^^^^^^^

error: invalid register `r2`: unknown register
    --> nrf-softdevice-s140/src/bindings.rs:1739:9
```

...and many more. I've also tried enabling features:

```
cargo run --features s140,nrf52840,ble-central,ble-peripheral,ble-l2cap,ble-gatt-client,ble-gatt-server --bin ble_bas_peripheral
```

Dirbaio, 2020-12-26:
You need to run the `cargo run` from the `examples/` dir, so the `target` from `.cargo/config` gets applied. 

Will update README now, sorry about the confusion! :)

huntc, 2020-12-26:
No problem! I'm happy to raise PRs also on this - and was going to include another section for ensuring the build environment is correct.

huntc, 2020-12-26:
Next error:

```
    Finished dev [optimized + debuginfo] target(s) in 33.64s
     Running `probe-run --chip nRF52840_xxAA --defmt /Users/huntc/Projects/hacking/nrf-softdevice/target/thumbv7em-none-eabihf/debug/ble_bas_peripheral`
error: could not execute process `probe-run --chip nRF52840_xxAA --defmt /Users/huntc/Projects/hacking/nrf-softdevice/target/thumbv7em-none-eabihf/debug/ble_bas_peripheral` (never executed)

Caused by:
  No such file or directory (os error 2)
```

Dirbaio, 2020-12-26:
`cargo install probe-run` 

There's definitely lots of room for improvement in the README. PRs very welcome! :)

huntc, 2020-12-26:
A PR is coming - thanks for the help!

huntc, 2020-12-26:
How does `cargo run --bin ble_bas_peripheral` relate to the s140 and nRF52840? Fortunately for me, those are my defaults, and it works beautifully. However, I'd expect to be explicit about the soft device and hard device here.

Dirbaio, 2020-12-26:
It's configured from your crate, through Cargo features: https://github.com/akiles/nrf-softdevice/blob/master/examples/Cargo.toml#L35

huntc, 2020-12-26:
> It's configured from your crate, through Cargo features: https://github.com/akiles/nrf-softdevice/blob/master/examples/Cargo.toml#L35

OK, so to override then the expectation is to modify the `cargo.toml` on that line and below?

Dirbaio, 2020-12-26:
Yes. 

The examples crate could have features that are forwarded to the nrf-softdevice crate so that you can run the examples with any softdevice. With any chip too, but then you have to forward that to the PAC and HAL and memory.x...

It gets quite complicated quite fast, that's why I didn't do it. I'd rather keep examples simple. Instructions to run the example on other chips/softdevices could be in the readme.

PR #40 (2020-12-26): Smooth the getting started
Some changes to the README given guidance. Also, a benign addition to the VS Code settings to stop CMake from being annoying in the case where it is installed.

Fixes #38
closed 2020-12-26 by Dirbaio