embassy-rs/nrf-softdevice #47: Better logs on softdevice faults

Dirbaio, 2021-01-22:


fnafnio, 2021-01-23:
I've looked into softdevice faults and there doesn't seem to be a lot of information on the nordic docs regarding specific codes
https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v7.2.0/group___n_r_f___f_a_u_l_t___i_d___r_a_n_g_e_s.html

Is there more information to be found in the softdevice SDK/header files?

The suggestions based on the fault ID  sound reasonable, it would certainly increase the readability of the fault messages (and possibly stop people like me from creating issues that are their own errors 😐)

Printing a stack trace would be great, but wouldn't it be needed to get back to the user application stack and panicking there? Would that even be allowed when the softdevice already faulted?

Dirbaio, 2021-01-23:
> Is there more information to be found in the softdevice SDK/header files?

Not much, just this:

https://github.com/akiles/nrf-softdevice/blob/03c46e7b4260985033ae0e5323b532350bfa77b8/softdevice/s140/headers/nrf_sdm.h#L177-L182
https://github.com/akiles/nrf-softdevice/blob/03c46e7b4260985033ae0e5323b532350bfa77b8/softdevice/s140/headers/nrf_sdm.h#L270-L290

> wouldn't it be needed to get back to the user application stack and panicking there? Would that even be allowed when the softdevice already faulted?

Yeah, that wouldn't be allowed. I was hoping to get probe-run to unwind through softdevice frames to get back to user code, but I'm not sure hos feasible that is.

Dirbaio, 2021-02-23:
Done in 929ce945ef0

closed 2021-02-23 by Dirbaio