embassy-rs/nrf-softdevice #49: How to enable PHY_CODED?

rise0chen, 2021-01-28:


Dirbaio, 2021-01-28:
For peripheral role (advertising): set `primary_phy` and `secondary_phy` to Phy::Coded in `peripheral::Config`.

For central role (connecting) it's currently hardcoded. Sorry about that! :( A PR would be very welcome!

https://github.com/akiles/nrf-softdevice/blob/03c46e7b4260985033ae0e5323b532350bfa77b8/nrf-softdevice/src/ble/central.rs#L81

If you want maximum range, also set `tx_power: TxPower::Plus8dBm` (for nrf52840 +8dbm, I think the other chips support only up to +4dbm?)

rise0chen, 2021-01-28:
```
WARN [nrf_softdevice::ble::peripheral] sd_ble_gap_adv_set_configure err NotSupported
WARN [nrf_softdevice::ble::peripheral] sd_ble_gap_adv_stop: BleInvalidAdvHandle
ERROR [btn::common::logger] panicked at 'fault_handler 1 92876 0', C:\Users\admin\.cargo\git\checkouts\nrf-softdevice-9b99539d60cc72a7\f4b2e62\nrf-softdevice\src\softdevice.rs:13:5
```

panic when i set `peripheral::Config.primary_phy = Phy::Coded`
```
 peripheral::Config {
        primary_phy: Phy::Coded,
        secondary_phy: Phy::_1M,
}
```

Dirbaio, 2021-01-28:
Coded PHY only supports extended advertisements, so you have to use `ConnectableAdvertisement::ExtendedNonscannableUndirected`

Aditionally I think you have to set both primary and secondary phy to Coded. The secondary phy is what the final connection will use, you probably want Coded for it. Softdevice docs say this about it:

    If @ref ble_gap_adv_properties_t::type is an extended advertising type
    and connectable, this is the PHY that will be used to establish a
    connection and send AUX_ADV_IND packets on.

rise0chen, 2021-01-29:
The question remains, when use `ConnectableAdvertisement::ExtendedNonscannableUndirected`
```rust
let config = peripheral::Config {
    primary_phy: Phy::Coded,
    secondary_phy: Phy::Coded,
    tx_power: TxPower::Minus4dBm,
    timeout: None,
    max_events: None,
    interval: 3200, //2s
};
let adv = peripheral::ConnectableAdvertisement::ExtendedNonscannableUndirected { adv_data: data };
let conn = (peripheral::advertise(sd, adv, &config).await).unwrap();
```

maybe, Softdevice don't support Coded PHY?

Dirbaio, 2021-01-29:
Coded phy is only supported on nrf52811, nrf52820, nrf52833, nrf52840, and only on softdevice S140. Are you using one of these?

rise0chen, 2021-01-29:
oh,i am using 52832.

closed 2021-01-29 by rise0chen