embassy-rs/nrf-softdevice #59: Update examples to work with latest embassy

Dirbaio, 2021-04-06:
Latest embassy always requires a clock to be set (this itself may be an embassy bug, have to look into it).

Either way, the examples should Just Work and they don't right now. They're missing manual clock setup like the following, or a way to set the irq priority with `#[embassy::main]`

```rust

    unsafe { embassy_nrf::system::configure(Default::default()) };

    let irq = interrupt::take!(RTC1);
    irq.set_priority(Priority::Level3); // levels 0-1 are reserved for the softdevice
    let rtc = unsafe { embassy_nrf::peripherals::RTC1::steal() };
    let rtc = RTC.put(rtc::RTC::new(rtc, irq));
    rtc.start();
    unsafe { embassy::time::set_clock(rtc) };

chris-ricketts, 2021-05-26:
Getting the following in the ble_peripheral_onoff example after #67 / #66 were merged:
```rust
       0 INFO  Hello World!
└─ raw_test::__cortex_m_rt_main @ src/bin/raw-test.rs:135
       1 ERROR panicked at 'sd_softdevice_enable err SdmIncorrectInterruptConfiguration'
└─ nrf_softdevice::softdevice::{{impl}}::enable @ /home/chris/.cargo/git/checkouts/nrf-softdevice-03ef4aef10e777e4/17c8983/nrf-softdevice/src/softdevice.rs:111
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:172
   2: __udf
        at ./asm/lib.rs:49
   3: rust_begin_unwind
        at /home/chris/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.1.0/src/lib.rs:43
error: the stack appears to be corrupted beyond this point
```
Seems to be when the `embassy_nrf::init(...)` function is used. 

This is on a nrf52840.

Dirbaio, 2021-05-26:
@chris-ricketts Fixed in master! :) 

That was due to `embassy_nrf::init()` enabling the RTC interrupt at too high priority.

chris-ricketts, 2021-05-26:
@Dirbaio thanks a lot! That example works really well now.

I have a couple of questions on how to enable the use of `embassy::time::Timer` when using `nrf-softdevice` and also the right way to take ownership of peripherals that are not used by `embassy_nrf::Peripherals`, e.g. USBD.
 
Should I make separate issues for these questions with 'Question' in the title? Might help others.

chris-ricketts, 2021-05-26:
Answered my own question RE: `embassy::time::Timer` - you simply need to set it up the same way as in the `embassy-nrf` `raw_spawn.rs` example:
```rust
static RTC: Forever<rtc::RTC<peripherals::RTC1>> = Forever::new();
static ALARM: Forever<rtc::Alarm<peripherals::RTC1>> = Forever::new();
static EXECUTOR: Forever<Executor> = Forever::new();

#[embassy::task]
async fn timer_task() {
    loop {
        info!("tick");
        Timer::after(Duration::from_ticks(13000)).await;
    }
}

#[embassy::task]
async fn softdevice(sd: &'static Softdevice) {
    sd.run().await;
}

#[entry]
fn main() -> ! {
  let mut nrf_config = embassy_nrf::config::Config::default();
  config.gpiote_interrupt_priority = Priority::P2;
  let p = embassy_nrf::init(nrf_config);
  
  let rtc = RTC.put(rtc::RTC::new(p.RTC1, interrupt::take!(RTC1)));
  rtc.start();
  unsafe { embassy::time::set_clock(rtc) };

  let alarm = ALARM.put(rtc.alarm0());
  let executor = EXECUTOR.put(Executor::new());
  executor.set_alarm(alarm);

  let sd_config = nrf_softdevice::Config { ... };
  let sd = Softdevice::enable(&sd_config);

  executor.run(|spawner| {
    unwrap!(spawner.spawn(softdevice(sd)));
    unwrap!(spawner.spawn(timer_task()));
  });
}
```

Dirbaio, 2021-05-27:
> the right way to take ownership of peripherals that are not used by embassy_nrf::Peripherals, e.g. USBD.

You can do `nrf52840_pac::Peripherals::take()` in parallel to embassy's and use both. You can also "steal" individual singletons with `let usb: nrf52840_pac::USBD = mem::transmute(());`.

In both cases you have to ensure you don't use the same peripheral with both HALs of course :) 

> Answered my own question RE: embassy::time::Timer

Nice! You'll want to set the prio on the irq with `let irq = interrupt::take!(RTC1); irq.set_priority(interrupt::Priority::P3);`. Prio 0 and 1 are reserved for the softdevice, and the default is 0.

> Should I make separate issues for these questions with 'Question' in the title? Might help others.

Question issues are welcome! If they're unrelated it's nicer to post them in their own issue.

Dirbaio, 2021-05-27:
Examples now work fine with latest embassy, closing.

closed 2021-05-27 by Dirbaio

kuon, 2021-08-16:
I am trying to use embassy with softdevice, but if I add this code:

```rust
    let mut config = embassy_nrf::config::Config::default();
    config.gpiote_interrupt_priority = Priority::P2;
    let p = embassy_nrf::init(config);

```

before `softdevice::enable` I have a panic:

```sh
     Running `probe-run --chip nRF52840_xxAA target/thumbv7em-none-eabihf/debug/sensor-firmware`
(HOST) INFO  flashing program (68.47 KiB)
(HOST) INFO  success!
────────────────────────────────────────────────────────────────────────────────
0 ERROR panicked at 'sd_softdevice_enable err SdmIncorrectInterruptConfiguration'
└─ nrf_softdevice::softdevice::{impl#0}::enable @ /home/kuon/.cargo/git/checkouts/nrf-softdevice-03ef4aef10e777e4/29211a5/nrf-softdevice/src/fmt.rs:101
1 ERROR panicked at 'explicit panic', /home/kuon/.cargo/registry/src/github.com-1ecc6299db9ec823/defmt-0.2.3/src/lib.rs:298:5
└─ panic_probe::print_defmt::print @ /home/kuon/.cargo/registry/src/github.com-1ecc6299db9ec823/panic-probe-0.2.0/src/lib.rs:94
────────────────────────────────────────────────────────────────────────────────
stack backtrace:
   0: HardFaultTrampoline
      <exception entry>
   1: lib::inline::__udf
        at ./asm/inline.rs:172:5
   2: __udf
        at ./asm/lib.rs:49:17
   3: cortex_m::interrupt::disable
        at /home/kuon/.cargo/registry/src/github.com-1ecc6299db9ec823/cortex-m-0.7.3/src/interrupt.rs:37
(HOST) WARN  call stack was corrupted; unwinding could not be completed
(HOST) ERROR the program panicked
make: *** [Makefile:16: run] Error 134
```

I am using git master version of embassy and softdevice. (cbff0398 for embassy and 29211a57 for softdevice)

Dirbaio, 2021-08-18:
@kuon latest `embassy-nrf` can now set the RTC1 irq prio with `config.time_interrupt_priority` :)

kuon, 2021-08-18:
Great! I wasn't sure that it was what was fixed in this thread. Thanks a lot.

Dirbaio, 2021-08-18:
Not really, the issue is something else:

- Back when this issue was filed, the time driver ("clock") setup was done manually so all that was necessary was to set the RTC1 IRQ prio. 
- Since embassy-rs/embassy#338 the time driver setup is done automatically in `embassy_nrf::init()`, so setting the prio was no longer possible.
- I've now added back the possibility to do that in `embassy-rs/embassy#364