embassy-rs/nrf-softdevice #64: Question: How to safely configure SoftDevice peripherals before giving up ownership?

chris-ricketts, 2021-05-05:
Hi,

First off, thanks for all this great work on embassy and nrf-softdevice! 

Sorry if this is a noob question.

I am trying to combine this crate with the [usb-device](https://github.com/mvirkkunen/usb-device). I have both working separately.

For the USB stuff I use [enable_ext_hfosc](https://docs.rs/nrf52840-hal/0.12.1/nrf52840_hal/struct.Clocks.html#method.enable_ext_hfosc) to configure the clock.

Something like this:
```rust
let periph = Peripherals::take().unwrap();
let clocks = nrf52840_hal::Clocks::new(periph.CLOCK);
let clocks = clocks.enable_ext_hfosc();
```
As `Softdevice::Peripherals` requires ownership of `CLOCK` I don't see a way to do this other than manually setting the register.

Dirbaio, 2021-05-05:
The softdevice takes over managing the clocks for you, so (unfortunately?) there's no way to use `nrf52840_hal::Clocks` anymore.

Instead, you have to ask the softdevice to enable HFCLK for you, with [`sd_clock_hfclk_request()`](https://github.com/embassy-rs/nrf-softdevice/blob/77db337aef97c58d0824b160778b100a31a46ebf/softdevice/s140/headers/nrf_soc.h#L698). See [here](https://devzone.nordicsemi.com/f/nordic-q-a/46640/hfclk-start-and-usage). A call with error checking would look like:

```rust
    unwrap!(RawError::convert(unsafe { raw::sd_clock_hfclk_request() }));
```

You'll probably want to loop polling `sd_clock_hfclk_is_running()` to wait for the hfclk to be started.

chris-ricketts, 2021-05-05:
Thanks a lot for the quick response @Dirbaio! 

That makes sense. 

A more ergonomic (and safe) API can be figured out at some point. I'll try and look into it but I'm still getting my bearings.

closed 2021-05-05 by chris-ricketts

Dirbaio, 2021-05-05:
Agreed re a nicer API! at the very least it should be documented!