embassy-rs/nrf-softdevice #79: Low power with softdevice

kuon, 2021-09-03:
This might be a bit of a broad question, but how can I make a gatt server low power?

I have an app similar to the `ble_bas_peripheral` example.

Basically, I have 3 tasks:

- softdevice
- bluetooth
- sensor read

The sensor read task is similar to this:
https://github.com/embassy-rs/embassy/blob/master/examples/nrf/src/bin/twim_lowpower.rs

And the sensor task has a channel to communicate with the bluetooth task to push data to it.

Dirbaio, 2021-09-03:
Yeah, lowpower should Just Work.

The embassy executor automatically puts the CPU to sleep with `WFE` when there's no work to do. 

If you ensure all peripherals that require HFCLK are disabled when not needed (like with that lowpower example) you should get super low power operation :)

kuon, 2021-09-03:
So, even if `sd.run().await` is still running, embassy will be able to put the device to sleep?

Dirbaio, 2021-09-03:
Yep!

closed 2022-02-05 by Dirbaio