embassy-rs/nrf-softdevice #97: Use of `cortex_m::interrupt::free` in examples

Stupremee, 2022-01-29:
In the readme you say that you should never use `cortex_m::interrupt::free` because of softdevice. However, in the examples the `alloc-cortex-m` crate is used, which uses `cortex_m::interrupt::free` in their implementation [here](https://github.com/rust-embedded/alloc-cortex-m/blob/6c9060ff803f5ac77b003c2cd7764cc3f1beaed2/src/lib.rs#L67-L80).

Dirbaio, 2022-02-05:
Oops, nice catch. This is not nice, indeed.

GATT examples can work without alloc as-is. Alloc is only used for managing l2cap buffers, I have a project where I do that with `atomic-pool` instead, it's quite nicer. I should update the examples to that (#69)

Stupremee, 2022-02-05:
Maybe you could add a section in the readme where you list some common crates that use `cortezx_m::interrupt::free`

alexmoon, 2022-09-16:
Closed by #69

closed 2022-09-16 by alexmoon