embassy-rs/nrf-softdevice #100: gatt: Panics when receiving writes with bad length

Dirbaio, 2022-02-11:
How to reproduce:
- Create a writable char of type `u8`.
- Connect to the device with nRF Connect, write an empty value to it.

I have no idea why this happens.

there's this `vlen` attr (for "variable length") https://github.com/embassy-rs/nrf-softdevice/blob/master/nrf-softdevice/src/ble/gatt_server.rs#L21
it's `false` if MIN_LEN==MAX_LEN: https://github.com/embassy-rs/nrf-softdevice/blob/master/nrf-softdevice-macro/src/lib.rs#L246
which is the case for something like `u8` https://github.com/embassy-rs/nrf-softdevice/blob/master/nrf-softdevice/src/ble/gatt_traits.rs#L39-L40

So either I'm misunderstanding what `vlen` does, or there's some bug somewhere (in our code or in the softdevice??)

xgroleau, 2022-06-29:
After some testing I can confirm that on pure C code, disabling `vlen` doesn't prevent variable length received on the server. So this is not a bug, simply a misunderstanding of what `vlen` does. If the client writes 0 bytes, the received packet is 0 on the server. 

Variable length only changes the length of  the client when he reads or is notified by the server, not when the server receives data. 

I don't know what would be the best way to handle this, but padding with zeroes (while considering endianness) could be an option.

chrysn, 2022-09-20:
Attempting to clarify to any other new user of this crate, because this took me a while running into the issue and finding this:

AIU, the softdevice should already filter out attempts to write outside of the bounds established by this crate. Some interpretation difference between the softdevice and this crate causes issues. At least from this crate's point of view, the many `unwrap!` calls in gatt_traits.rs are justified, just something about when the size guards should kick in is unclear.

(This doesn't change anything about this being an issue, but the crate authors are *not* lightly spreading unwraps in paths that are user controlled).

alexmoon, 2022-09-20:
I've created a quick fix at #139. Writes with a bad length will be ignored rather than panicking. If more sophisticated handling is needed you can implement the service by hand. Let me know if the PR resolves the issue for you.

PR #139 (2022-09-21): Fix panic on write with bad length
Fixes #100 for services generated with the `gatt_service` macro.
commit 1c6d3304 (2022-09-21): Fix panic on write with bad length

Fixes #100 for services generated with the `gatt_service` macro.
closed 2022-09-21 by Dirbaio