embassy-rs/nrf-softdevice #113: nrf51 series

deliciouslytyped, 2022-06-03:
Is there an alternative package for the nrf51 series, or how difficult would it be to support?

Dirbaio, 2022-06-03:
There's no crate for it out there AFAIK.

It shouldn't be too hard, mostly:
- Add headers for nrf51 softdevices in `/softdevice` (S110, S120, S130)
- Generate `nrf-softdevice-sXX` crates for them
- Add relevant cfg's to `nrf-softdevice` crate
- Fix whatever compilation errors pop up (I believe the nrf52 softdevice's API is mostly a superset of nrf51's, so this should be mostly cfg-ing out things?)

PRs welcome :)

alexmoon, 2022-09-16:
PRs are still welcome, but at this point I don't think there's any intent to support the nrf51 so I'm closing the issue.

closed 2022-09-16 by alexmoon