embassy-rs/nrf-softdevice #119: Support passing attribute in nrf-softdevice macros

xgroleau, 2022-06-29:
Allowing to propagate attribute could be helpful in the nrf-softdevice-macro create. This would allow to do conditional compilation or maybe ble documentation directly in the code.

As an example, this code will not compile if `dfu` is not enabled
```
#[nrf_softdevice::gatt_server]
pub struct Server {
    #[cfg(feature = "dfu")]
    pub dfu: DfuService,
}
```

alexmoon, 2022-07-15:
We've just added a Builder pattern API which you can use instead of the macros to get full control including conditional compilation. You can look at the new `ble_dis_bas_peripheral_builder` example or the proc macro expansion to see how to use the new API.

alexmoon, 2022-09-16:
The ServiceBuilder is an effective workaround for this issue.

closed 2022-09-16 by alexmoon