embassy-rs/nrf-softdevice #168: Including in a separate Cargo project

olalonde, 2023-01-19:
It would be helpful to have some documentation on how to use this from a separate project. I tried to move out the `examples/` but it doesn't build...

``` console 
$ cargo run --bin ble_bas_peripheral
    Updating crates.io index
error: failed to get `nrf-softdevice` as a dependency of package `nrf-softdevice-examples v0.1.0 (/Users/redacted/code/nordic/myapp)`

Caused by:
  failed to load source for dependency `nrf-softdevice`

Caused by:
  Unable to update /Users/redacted/code/nordic/nrf-softdevice

Caused by:
  found a virtual manifest at `/Users/redacted/code/nordic/nrf-softdevice/Cargo.toml` instead of a package manifest
```

I tried removing `path = "../nrf-softdevice"` and `path = "../nrf-softdevice-s140"` from `Cargo.toml` but I have more errors:

```console
$ cargo run --bin ble_bas_peripheral
    Updating crates.io index
error: failed to select a version for the requirement `embassy-nrf = "^0.1.0"`
candidate versions found which didn't match: 0.0.0
location searched: crates.io index
required by package `nrf-softdevice-examples v0.1.0 (/Users/redacted/code/nordic/myapp)`
```

Help appreciated.

sureshjoshi, 2023-01-25:
Not sure if it's still helpful, but here's what I use for my S340 fork. I'm only just learning Rust, so maybe there is a way to shorten this, but 🤷🏽 

```bash
[package]
name = "myproject"
version = "0.0.1"
description = "Firmware for myproject"
authors = [
    "SJ <me@sureshjoshi.com>"
]
edition = "2021"
publish = false

[features]
default = [
    "nightly",
    "ble-l2cap",
    "ble-gatt-server",
    "ble-gatt-client",
    "ble-sec",
]
nightly = ["embassy-executor/nightly", "embassy-nrf/nightly", "embedded-io/async"]

ble-l2cap = ["nrf-softdevice/ble-l2cap"]
ble-gatt-server = ["nrf-softdevice/ble-gatt-server"]
ble-gatt-client = ["nrf-softdevice/ble-gatt-client"]
ble-sec = ["nrf-softdevice/ble-sec"]

[dependencies]
embassy-futures = { git = "https://github.com/embassy-rs/embassy" }
embassy-executor = { git = "https://github.com/embassy-rs/embassy", features = ["defmt", "integrated-timers"] }
embassy-nrf = { git = "https://github.com/embassy-rs/embassy", features = ["defmt", "nrf52840", "time-driver-rtc1", "gpiote", "unstable-pac"] }
embassy-sync = { git = "https://github.com/embassy-rs/embassy", features = ["defmt"] }
embassy-time = { git = "https://github.com/embassy-rs/embassy", features = ["defmt", "defmt-timestamp-uptime"] }
embedded-io = "0.4.0"

nrf-softdevice = { git = "ssh://myprivatefork", features = ["defmt", "nrf52840", "s340", "ble-peripheral", "ble-central", "critical-section-impl"] }
nrf-softdevice-s340 = { git = "ssh://myprivatefork" }

defmt = "0.3"
defmt-rtt = "0.4"

cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.0"
panic-probe = { version = "0.3", features = ["print-defmt"] }
```

I'm sure instead of `nrf-softdevice-s340 = { git = "ssh://myprivatefork" }` you could do `nrf-softdevice-s340 = "0.1.1"`

materoy, 2023-03-11:
> 
Build with nightly version of rust, check rust-toolchain.toml

sureshjoshi, 2023-06-28:
@olalonde Did this ever work out for you?