embassy-rs/nrf-softdevice #222: Add descriptor to a characteristic

AliMoal, 2024-01-27:
How to add a descriptor (for example UUID: 0x2904) to a characteristic?
I used `add_descriptor` function but it returns: `Raw(Forbidden)` and the program panics.
If any example was provided, it would be more helpful.
Thanks.

alexmoon, 2024-01-28:
Per [Nordic's documentation](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v7.3.0/group___b_l_e___g_a_t_t_s___f_u_n_c_t_i_o_n_s.html?cp=5_7_4_1_2_4_2_2#ga0e1478bd565ad99f998671e3baf9847d) certain UUID values are reserved to the softdevice. The characteristic presentation format is one of those. Instead, you need to add the presentation format definition to the characteristic metadata. However, that is not currently implemented in our `Metadata` struct.

PRs are welcome.

AliMoal, 2024-01-30:
Thank you for your information.
I'll try it.

closed 2024-01-30 by AliMoal