embassy-rs/nrf-softdevice #224: NRF52832/S132 Freezes Upon Connection

AdinAck, 2024-01-30:
I have an nrf52832 (512K/64K) and ran the examples on it, but experienced a problem.

Advertisement works fine, so the SD is definitely up and running, but the moment a connection starts, it seems to just stop the main thread.

# Steps to Reproduce
1. Download s132 from nordic
2. Erase nrf
3. Download s132 to nrf with `probe-rs download --verify --format hex --chip nRF52840_xxAA ~/Downloads/s132_nrf52_7/s132_nrf52_7.3.0_softdevice.hex`
4. Change `memory-nrf52832.x` to allocate *31* KiB of RAM for SD
5. Run

Results in this output:
```
0.000000 INFO  Hello World!
└─ ble_bas_peripheral::____embassy_main_task::{async_fn#0} @ src/bin/ble_bas_peripheral.rs:42  
0.000000 INFO  softdevice RAM: 31480 bytes
└─ nrf_softdevice::softdevice::{impl#0}::enable @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:138 
0.000000 WARN  You're giving more RAM to the softdevice than needed. You can change your app's RAM start address to 20007af8
└─ nrf_softdevice::softdevice::{impl#0}::enable @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:151 
0.000000 TRACE ble evt 16
└─ nrf_softdevice::ble::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 DEBUG conn_params conn_sup_timeout=72 max_conn_interval=24 min_conn_interval=24 slave_latency=0
└─ nrf_softdevice::ble::gap::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:125 
0.000000 DEBUG connected role=Peripheral peer_addr=RandomPrivateResolvable:[7e, 5b, a6, 71, 94, 7f]
└─ nrf_softdevice::ble::peripheral::advertise_inner::{async_fn#0}::{closure#1} @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:125 
0.000000 TRACE conn 0: connected
└─ nrf_softdevice::ble::connection::{impl#9}::new::{closure#0} @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 INFO  advertising done!
└─ ble_bas_peripheral::____embassy_main_task::{async_fn#0} @ src/bin/ble_bas_peripheral.rs:101 
0.000000 TRACE ble evt 33
└─ nrf_softdevice::ble::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 TRACE on_phy_update_request conn_handle=3 rx_phys=2 tx_phys=2
└─ nrf_softdevice::ble::gap::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 TRACE ble evt 36
└─ nrf_softdevice::ble::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 DEBUG on_data_length_update conn_handle=3 max_rx_octets=251 max_rx_time_us=2120 max_tx_octets=251 max_tx_time_us=2120
└─ nrf_softdevice::ble::gap::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:125 
0.000000 TRACE ble evt 34
└─ nrf_softdevice::ble::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 TRACE on_phy_update conn_handle=3 status=0 rx_phy=2 tx_phy=2
└─ nrf_softdevice::ble::gap::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 TRACE ble evt 85
└─ nrf_softdevice::ble::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 TRACE att mtu exchange: peer wants mtu 527, granting 256
└─ nrf_softdevice::ble::gatt_server::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 TRACE ble evt 80
└─ nrf_softdevice::ble::on_evt @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
0.000000 TRACE gatts write handle=13 data=[2, 0]
└─ nrf_softdevice::ble::gatt_server::run::{async_fn#0}::{closure#1} @ /Users/adinack/Projects/nrf-softdevice/nrf-softdevice/src/fmt.rs:112 
```

This looks like the usual sequence of traces when accepting a connection, but the connection is never established, and no more traces are printed after this, no matter what the client does.

I tried running a blinking LED task in the background, that continued to blink.

I've been using this repo for months now on an nrf52840, but it has been a while since I first configured it, so maybe I'm missing something, nonetheless, if I am, there should be documentation explaining this behavior.

Thanks.

plaes, 2024-01-31:
I tried with my own custom nrf52832-based device and the same `ble_bas_peripheral` example seems to be working with both debug and release profile.

One discrepancy I noticed in your 3rd step is the usage of wrong chip name used: `--chip nRF52840_xxAA`.
Did you also change the  `runner` variable in `.cargo/config.toml`?

AdinAck, 2024-01-31:
This has been resolved, somehow `nrfjprog --recover` makes it all work, I have no explanation as to why.

closed 2024-01-31 by AdinAck