embassy-rs/nrf-softdevice #228: There is currently no way to disable the SoftDevice

mini-ninja-64, 2024-02-10:
Disabling the SoftDevice is useful as it allows for runtime configuration changes of the BLE stack, as config cannot be changed while the BLE SoftDevice is enabled.

> [sd_ble_cfg_set](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v7.3.0/group___b_l_e___c_o_m_m_o_n___f_u_n_c_t_i_o_n_s.html#ga4edae2bac8c68b672c0fa101ed2c687f) may be called at any time when the SoftDevice is enabled (see [sd_softdevice_enable](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v7.3.0/group___n_r_f___s_d_m___f_u_n_c_t_i_o_n_s.html#ga420235cbdf6b0746ce86af3cfd0ec055)) while the BLE part of the SoftDevice is not enabled (see [sd_ble_enable](https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v7.3.0/group___b_l_e___c_o_m_m_o_n___f_u_n_c_t_i_o_n_s.html#ga8a92420df90945526d4c89d09486508d)).

[source](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s132.api.v4.0.2%2Fgroup___b_l_e___c_o_m_m_o_n___f_u_n_c_t_i_o_n_s.html)

I am still fairly new to Rust & the nRF5 series in general but can try and raise a PR for this (ﾉ◕ヮ◕)ﾉ*:･ﾟ✧

I am not sure what the API should look like for this?
maybe like `pub async fn disable(&self)` on `SoftDevice` with either noop or panic if its not currently enabled :thinking:.

alexmoon, 2024-02-11:
Unfortunately it’s not possible to disable the softdevice without introducing UB because the Softdevice is returned as a static reference. The current recommendation if you need to reconfigure the softdevice is to restart the microcontroller. It is pretty rare in practice for this the be an issue.

mini-ninja-64, 2024-05-26:
that makes sense, thank you, i have learned more from when i raised this issue and now understand :bowing_woman:

closed 2024-05-26 by mini-ninja-64