embassy-rs/nrf-softdevice #233: Softdevice peripheral clash with nRF applications

plaes, 2024-02-21:
I'm currently trying to set up an application with nrf52832 which uses both softdevice and LoRa and there seems to be some kind of timer-related clash happening regarding timers:

`Softdevice memory access violation. Your program accessed registers for a peripheral reserved to the softdevice. PC=2a644 PREGION=8192`

Relevant code in my application seems to be setting up the "embassy timer": https://github.com/lora-rs/lora-rs/blob/6895fcd227b44cf5359801d96527f2fe4b1b432e/lorawan-device/src/async_device/embassy_time.rs#L11-L14
```
   2a624:       f011 fce7       bl      3bff6 <_ZN14lorawan_device12async_device12embassy_time12EmbassyTimer3new17h35720142ac53d592E>
   2a628:       4605            mov     r5, r0
   2a62a:       f24d 0004       movw    r0, #53252      @ 0xd004
   2a62e:       f2c4 0000       movt    r0, #16384      @ 0x4000
   2a632:       f04f 0801       mov.w   r8, #1
   2a636:       f8c0 8000       str.w   r8, [r0]
   2a63a:       460e            mov     r6, r1
   2a63c:       f8c0 8304       str.w   r8, [r0, #772]  @ 0x304
   2a640:       f44f 5000       mov.w   r0, #8192       @ 0x2000    // <<<
   2a644:       f8cb 0180       str.w   r0, [fp, #384]  @ 0x180
   2a648:       f8cb 0000       str.w   r0, [fp]
   2a64c:       f104 0048       add.w   r0, r4, #72     @ 0x48
   2a650:       4651            mov     r1, sl
   2a652:       220e            movs    r2, #14
   2a654:       2300            movs    r3, #0
   2a656:       f04f 0900       mov.w   r9, #0
```

Dirbaio, 2024-02-21:
yep it's RNG. If you check the ASM it's writing to 0x4000d004 which is a RNG reg.

You have to use this instead of RNG directly https://docs.rs/nrf-softdevice/latest/nrf_softdevice/fn.random_bytes.html

plaes, 2024-02-21:
> yep it's RNG. If you check the ASM it's writing to 0x4000d004 which is a RNG reg.
> 
> You have to use this instead of RNG directly https://docs.rs/nrf-softdevice/latest/nrf_softdevice/fn.random_bytes.html

Thanks, I'll cook up a PR for some README improvements.

closed 2024-02-21 by plaes

alexmoon, 2024-02-21:
@plaes Also note that LoRa and the softdevice both need exclusive access to the radio peripheral, so there's really no way to use them together in the same firmware.