embassy-rs/nrf-softdevice #235: Implement `NorFlash` for `Flash`

HaoboGu, 2024-02-27:
In current flash implementation https://github.com/embassy-rs/nrf-softdevice/blob/master/nrf-softdevice/src/flash.rs, the struct `Flash` implements only blocking `ReadNorFlash`, and async version of `ReadNorFlash` and `NorFlash`. Is there any concern to not implement blocking `NorFlash`?

Dirbaio, 2024-02-27:
the softdevice api is only async (you request the write/erase operation, you get an event later notifying it was done). We can't spin waiting for the operation to be done, because it'll starve the softdevice task so the event won't get processed. Same if you try to use `block_on()` from your side to convert the async api to blocking.

HaoboGu, 2024-02-27:
gotcha! thanks for the explanation:D

closed 2024-02-27 by HaoboGu