embassy-rs/nrf-softdevice #241: Could not execute process probe-run --chip nRF52840_xxAA

tomvictor, 2024-03-05:
I keep getting the same error, unable to find the exact root cause. Appreciate any suggestion to move forward !

```
cargo run --bin ble_bas_peripheral --features nrf52840
   Compiling nrf-softdevice-examples v0.1.0 (/Users/tomvictor/git/tom/embassy/ble/examples)
    Finished dev [unoptimized + debuginfo] target(s) in 0.25s
     Running `probe-run --chip nRF52840_xxAA /Users/tomvictor/git/tom/embassy/ble/target/thumbv7em-none-eabihf/debug/ble_bas_peripheral`
error: could not execute process `probe-run --chip nRF52840_xxAA /Users/tomvictor/git/tom/embassy/ble/target/thumbv7em-none-eabihf/debug/ble_bas_peripheral` (never executed)

Caused by:
  No such file or directory (os error 2)
```

Thanks

Dirbaio, 2024-03-05:
`cargo install probe-run`

Dirbaio, 2024-03-05:
actually the issue is the README tells you to install `probe-rs`, but the Cargo runner was still set to `probe-run`. I've fixed this in https://github.com/embassy-rs/nrf-softdevice/commit/182605bd9e5b90fa1a34534dd756a2fd674c1ab7 . Please update your git checkout and follow the instructions in the README, they should work now.

tomvictor, 2024-03-05:
Thanks @Dirbaio  for the quick reply and the fix. I really appreciate your efforts.

closed 2024-03-05 by tomvictor