embassy-rs/nrf-softdevice #246: Flash Capacity Definition Discrepancy for nRF52832 and nRF52833 Chips

lonesometraveler, 2024-03-23:
Thank you for maintaining this project.

I wanted to reach out to share an observation I made while using the flash module. It seems there might be an oversight in how the flash memory capacity is defined.

https://github.com/embassy-rs/nrf-softdevice/blob/e769274bf6812c55fb7e9d9f9e87b23a141b8b7a/nrf-softdevice/src/flash.rs#L87C1-L89C6

The library identifies the flash capacity as 1MB (256 * 4096), which is correct for nRF52840. However, for nRF52832 and nRF52833, the flash size is 512KB. Shouldn't it return different values based on the chip to accurately reflect their flash capacities?

alexmoon, 2024-03-25:
Yes, that's just an oversight. PR to fix would be welcome. Though I would note that capacity is not a terribly useful figure for the internal flash since you can't use it all without nuking your firmware.

plaes, 2024-04-09:
Not only flash sizes, but also page size of flash. And to make things complicated, nrf52832 has actually two variants: 512KB and 256KB variant.
So, while I'm at it here's a list of NRF devices and their flash sizes:
* nrf5340 - 1 MB + 256 KB 
* nRF52840 - 1 MB
* nRF52833 - 512 KB
* nRF52832 - 512/256 KB
* nRF52820 - 256 KB
* nRF52811 - 192 KB
* nRF52810 - 192 KB
* nRF52805 - 192 KB

Unfortunately I didn't find a definition for flash page size from SDK at quick glance.

alexmoon, 2024-04-09:
I believe they all have 4k pages, with the exception of the 5340 network core which has 2k pages.

commit 969d8872 (2024-04-30): flash: Return flash capacity per-chosen device

Previous value was hardcoded to return nrf52840's flash size.
Fixes #246.
PR #252 (2024-04-30): flash: Return flash capacity per-chosen device
Previous value was hardcoded to return nrf52840's flash size. Fixes #246.
closed 2024-04-30 by Dirbaio