embassy-rs/nrf-softdevice #258: nRF52805 example

standarddeviant, 2024-06-22:
Hello,

Firstly, thank you so much for embassy and nrf-softdevice. Being able to program microcontrollers with rust is very appealing.

I'm curious if anyone has successfully run a an embassy/nrf-softdevice BLE example on an nrf52805. I'm starting to work through how to do it by editing the `Cargo.toml` in the examples folder - but wondering if there are relevant examples or blog posts that I've missed.

plaes, 2024-06-27:
Well, you first look which softdevice is supported, and then check the documentation for the properties of this version, for example S113's softdevice Flash size is 112.0 kB (0x1C000 bytes).

plaes, 2024-08-13:
@standarddeviant You can follow these two pull requests to see how support for other nrf variants was added:
* nrf52833: #270 
* nrf52832: #211