embassy-rs/nrf-softdevice #260: Help with compiling - missing crates

Gibbz, 2024-06-30:
I'm trying to add the softdevice to my project. However I'm getting 118 build errors with the softdevice package. It seems to be missing crates, Ive tried adding them with "cargo add", but it doesn't work.

Here is my pastebin:
https://pastebin.com/1RpeFhBj

alexmoon, 2024-07-01:
You need to enable one of the "sXXX" features (e.g. "s140") on the `nrf_softdevice` crate in you Cargo.toml.

Gibbz, 2024-07-02:
Great thanks! its compiling now!

For others with the issue, my Cargo.toml
```
[features] 
default = [
    "ble-l2cap",
    "ble-gatt-server",
    "ble-gatt-client",
    "ble-sec",
]

ble-l2cap = ["nrf-softdevice/ble-l2cap"]
ble-gatt-server = ["nrf-softdevice/ble-gatt-server"]
ble-gatt-client = ["nrf-softdevice/ble-gatt-client"]
ble-sec = ["nrf-softdevice/ble-sec"]

nrf52840 = [
  "embassy-nrf/nrf52840",
  "nrf-softdevice/nrf52840",
  "nrf-softdevice/s140",
  "dep:nrf-softdevice-s140"
]

[dependencies]
# embassy
embassy-futures = { version = "0.1.0"}
embassy-sync = { version = "0.6.0", features = ["defmt"] }
embassy-executor = { version = "0.5.0", features = ["task-arena-size-32768", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"] }
embassy-time = { version = "0.3.1", features = ["defmt", "defmt-timestamp-uptime"] }
embassy-nrf = { version = "0.1.0", features = ["defmt", "nrf52840", "time-driver-rtc1", "gpiote", "unstable-pac", "time"] }

# log
defmt = "0.3.5"
defmt-rtt = "0.4"

# arm
cortex-m = { version = "0.7.7", features = ["inline-asm", "critical-section-single-core"] }
cortex-m-rt = "0.7.3"

# misc
panic-probe = { version = "0.3.1", features = ["print-defmt"] }
serde = { version = "1.0.136", default-features = false }

# device driver
mcp4725 = "0.4.2"
ads1x1x = "0.2.2"

# maths
num-traits = { version = "0.2.19", default-features = false, features = ["libm"] }

# nrf softdevice bluetooth
nrf-softdevice-s140 = { version = "0.1.2", optional = true }
nrf-softdevice = { version = "0.1.0", features = ["nrf52840", "s140", "defmt", "ble-peripheral", "ble-central", "critical-section-impl"] }
embedded-storage = "0.3.1"
embedded-storage-async = "0.4.1"
futures = { version = "0.3.29", default-features = false }
fixed = "1.24.0"
heapless = "0.8.0"
atomic-pool = "1.0.1"
static_cell = "2.0.0"
```

Gibbz, 2024-07-02:
Sorry, im still having an issue with this.
When using the code from the examples, we need to use a core crate.
```
use core::mem;
```

When i use this, I cannot compile, i get the single error:
```
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/bronson/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/bronson/.cargo/bin:/home/bronson/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/android-ndk:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/bronson/Android/Sdk/tools:/home/bronson/Android/Sdk/platform-tools" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustcAaCM6B/symbols.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.11he2q5fndgbefdn.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.11x5xt1wt1w3geix.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.12fnivugkzayoce8.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.12ypu4koi8im4qo7.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.134qhqn60eds5awr.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.15v66278v4ib66b4.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.161xgav8hif3nsp0.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.16k0dfkqef09qrd.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.16oyf6al5ly3g4sj.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.174kj06h2y6kd52m.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.18h4tavfoltu38ok.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1biw5fkm2cblbiig.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1dtyovcepbfjvw5z.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1e7t7xpmzox8wfmu.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1ewgbvayu0qogy46.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1fcg6nxizfmfl3y9.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1fyf49rfxdju2w0j.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1g2481hdd613edyk.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1hb21wsydo5z8gr3.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1hc5kzl7jepodbah.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1o8dg0bvtk546yub.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1odxvqfzcw449ciw.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1pig5e1ystru4612.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1qsmc4uc2wpnraeg.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1r998d7e9q2z5a3c.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1so33rxj278yhsu1.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1spdegpngegpvd4r.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1sy0mttfij2rkibk.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.1u4k51qjgz76a5jg.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.20lvus9jo8co46nj.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.27mr87rj86bf4ybr.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb49a.28z4yakohyh4lcsf.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-2b316ff02b3bb4 [... cut ...]
          >>>            nrf_softdevice-f300b21eb286c83f.nrf_softdevice.a60eb0a3c2e4f353-cgu.4.rcgu.o:(_critical_section_1_0_release) in archive /home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/libnrf_softdevice-f300b21eb286c83f.rlib
          >>> defined at lib.rs:285 (/home/bronson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:285)
          >>>            cortex_m-fb97e8f95d8bc989.cortex_m.9332e1ad4daf93ef-cgu.0.rcgu.o:(.text._critical_section_1_0_release+0x1) in archive /home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/libcortex_m-fb97e8f95d8bc989.rlib
```

alexmoon, 2024-07-02:
You need to remove the `critical-section-single-core` feature from `cortex-m`. It is not compatible with the softdevice.

Gibbz, 2024-07-03:
When i remove the ```critical-section-single-core`` I get the following error:

```

error: linking with `rust-lld` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/bronson/.rustup/toolchains/stable-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/bronson/.cargo/bin:/home/bronson/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/bin:/opt/android-ndk:/usr/lib/jvm/default/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin:/home/bronson/Android/Sdk/tools:/home/bronson/Android/Sdk/platform-tools" VSLANG="1033" "rust-lld" "-flavor" "gnu" "/tmp/rustckr8dN0/symbols.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.14gfil6cg0fziylu.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1c4klxpay6lt73wq.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1cffb8r3y2ur10q3.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1ctiw2mz8fa2wtyi.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1cty80ucs8s7du29.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1cx10t5suo6vwv5d.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1dxzbmsx888pkvne.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1iobgxo1ekaouna4.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1irjn8zso0e5ft1m.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1is79plj3t60gi2w.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1kmyapo5p789dfrr.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1nggjod5va1h5290.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1nopimm4rhjxy84o.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1nzckum0pkgskxws.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1p5wxuvjo5u35241.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1pm695fra8tk7wtd.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1vb80cw0aavxrlfe.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.1wfxefnin0kxqjn8.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.20rh8rjn54eilvgr.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.20z1i1l3qj48pgtc.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.22xhiclgk1fn3irs.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.23qpino3glf1i7li.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.24eqiiklne4jh307.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.29mf4y5h5gfmbil.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2balghlcl97gs2pv.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2d8wn8imw31rsipk.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2dtfvygq1ixv6pq8.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2nhwaw4jwh801a4i.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2nodfmhrajsckp7u.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2owi7dy245u7ezu7.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2rfzphvrlh2ibi5x.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2t9eux3sbi58hw30.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/bike_aid_nrf_rs-252983d24a92fe42.2untswagd9lrwd5o.rcgu.o" "/home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/dep [... cut ...] 0fce38922a9b94b.rlib
          >>> referenced by lib.rs:200 (/home/bronson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:200)
          >>>               embassy_sync-e59d5ff37768876c.embassy_sync.c51c9d5126782167-cgu.0.rcgu.o:(core::ptr::drop_in_place$LT$critical_section..with..Guard$GT$::h123902211f7fe3f8) in archive /home/bronson/Projects/bike-aid/bike-aid-nrf-rs/target/thumbv7em-none-eabi/debug/deps/libembassy_sync-e59d5ff37768876c.rlib
          
```

Gibbz, 2024-07-03:
Ok ive got it compiling!
Now when i run the example on the nrf52840 chip, I get the following error:
```
ERROR panicked at 'sd_softdevice_enable err SdmIncorrectInterruptConfiguration'
└─ nrf_softdevice::softdevice::{impl#0}::enable @ /home/bronson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/nrf-softdevice-0.1.0/src/fmt.rs:101 
ERROR panicked at /home/bronson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.8/src/lib.rs:367:5:
explicit panic
└─ panic_probe::print_defmt::print @ /home/bronson/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.2/src/lib.rs:104 
```

Should i open a separate issue for this?

daagaak, 2024-07-03:
Using the softdevice requires setting up embassy-nrf to adhere to the soft device interrupt requirements:
https://github.com/embassy-rs/nrf-softdevice/blob/master/examples/src/bin/ble_peripheral_onoff.rs#L82

Interrupt levels 0, 1 and 4 are reserved by the softdevice if it is enabled:
https://infocenter.nordicsemi.com/topic/sds_s140/SDS/s1xx/processor_avail_interrupt_latency/exception_mgmt_sd.html

The default is zero (I think) and so without further configuration at start up you will hit this assertion in the softdevice.

Gibbz, 2024-07-03:
Thanks. I have added the config to my main.rs
```
    let mut config = embassy_nrf::config::Config::default();
    config.gpiote_interrupt_priority = Priority::P2;
    config.time_interrupt_priority = Priority::P2;
    let p = embassy_nrf::init(config);
```

But im still getting the same issue when running.
Do I need to change the bindinterrupts im using for my i2c in some way?
```
    let i2c_bus = {
        use embassy_nrf::{bind_interrupts, peripherals::{self}};
        bind_interrupts!(struct Irqs {SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0 => twim::InterruptHandler<peripherals::TWISPI0>;});
        let config = twim::Config::default();
        let i2c = Twim::new(p.TWISPI0, Irqs, p.P0_06, p.P0_08, config); // sda: p0.06, scl: p0.08
        let i2c_bus = Mutex::new(i2c);
        //let i2c_bus = NoopMutex::new(RefCell::new(i2c));
        //let i2c_bus = embassy_sync::mutex::Mutex::new(RefCell::new(i2c));
        I2C_BUS.init(i2c_bus)
    };
```

If i remove the i2c device from code, I get this error when running:
```
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
ERROR probe_rs::architecture::arm::core::armv7m: The core is in locked up status as a result of an unrecoverable exception
Error: The core is locked up.
```

Gibbz, 2024-07-06:
I found the following to try and set priority on my interrupts, but it doesn't work:
https://github.com/embassy-rs/embassy/issues/1541

Example
```
    use embassy_nrf::interrupt::Interrupt;
    embassy_nrf::interrupt::TEMP::set_priority(Priority::P2);  // <- Error: failed to resolve: `TEMP` is a variant, not a module
    bind_interrupts!(struct Irqs {
        TEMP => temp::InterruptHandler;
    });
```

The same for twim
```
        use embassy_nrf::interrupt::Interrupt;
        embassy_nrf::interrupt::SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0::set_priority(interrupt::Priority::P2); // <- Error: failed to resolve: `...` is a variant, not a module
        bind_interrupts!(struct Irqs {SPIM0_SPIS0_TWIM0_TWIS0_SPI0_TWI0 => twim::InterruptHandler<peripherals::TWISPI0>;});
```

Gibbz, 2024-07-09:
Ok I fixed the priority issue. Now I just have the core lockup issue. I'll open a new issue for that.

closed 2024-07-09 by Gibbz