embassy-rs/nrf-softdevice #277: advertisement exceeded buffer length

Gibbz, 2024-10-25:
Im trying to add fast pair support to my device. However when adding any data I get the exceeded buffer length panic.
It works without the .services_16... line. But I add it I can no longer build.
How can I increase the buffer?

```rust
static ADV_DATA: LegacyAdvertisementPayload = LegacyAdvertisementBuilder::new()
        .flags(&[Flag::GeneralDiscovery, Flag::LE_Only])
        .services_16(ServiceList::Complete, &[ServiceUuid16::from_u16(0xFE2C)]) // fast pair
        .services_128(ServiceList::Incomplete, &[[0x9E, 0xCA, 0xDC, 0x24, 0x0E, 0xE5, 0xA9, 0xE0, 0x93, 0xF3, 0xA3, 0xB5, 0x01, 0x00, 0x40, 0x6E]]) // UART Service
        .short_name("BScooter")
        .build();
```

alexmoon, 2024-10-25:
Bluetooth legacy advertisements are limited to 31 bytes. You’ll need to move some data to your scan payload or remove it altogether.

Gibbz, 2024-10-25:
Ah! So scan is basically extended advertisement?

Gibbz, 2024-10-28:
Okay cheers. I've moved data to the scan data and it's working. I'll try play with restarting advertising too 👍

closed 2024-10-28 by Gibbz