embassy-rs/nrf-softdevice #279: Cannot connect to two peripheral

HaoboGu, 2024-11-15:
Hello, I'm using a BLE central to connect multiple peripherals. I set two addresses in whitelist in `ConnectConfig`:

```rust
    let addrs = &[
        &Address::new(AddressType::RandomStatic, peripheral_addr),
        &Address::new(AddressType::RandomStatic, peripheral_addr2),
    ];
    let mut config: central::ConnectConfig<'_> = central::ConnectConfig::default();
    config.scan_config.whitelist = Some(addrs);
```

Then I use the same `ConnectConfig` to connect the two peripherals:

```rust
let conn1_fut = async {
    let conn = match central::connect(sd, &config).await {
            Ok(conn) => conn,
            Err(e) => {
                error!("BLE peripheral connect error: {}", e);
                continue;
            }
        };
    conn
};
let conn2_fut = async {
    let conn = match central::connect(sd, &config).await {
            Ok(conn) => conn,
            Err(e) => {
                error!("BLE peripheral connect error: {}", e);
                continue;
            }
        };
    conn
};
let (conn1, conn2) = join(conn1_fut, conn2_fut).await
```

But the softdevice reports:

```
ERROR BLE peripheral connect error: Raw(BleGapWhitelistInUse)
└─ rmk::split::nrf::central::run_ble_client::{async_fn#0} @ /Users/haobogu/Projects/keyboard/rmk/rmk/src/split/nrf/central.rs:67  
WARN  sd_ble_gap_whitelist_set err BleGapWhitelistInUse
└─ nrf_softdevice::ble::central::{impl#5}::to_raw @ /Users/haobogu/.cargo/git/checkouts/nrf-softdevice-03ef4aef10e777e4/d5f023b/nrf-softdevice/src/fmt.rs:151 
ERROR BLE peripheral connect error: Raw(BleGapWhitelistInUse)
└─ rmk::split::nrf::central::run_ble_client::{async_fn#0} @ /Users/haobogu/Projects/keyboard/rmk/rmk/src/split/nrf/central.rs:67  
WARN  sd_ble_gap_whitelist_set err BleGapWhitelistInUse
└─ nrf_softdevice::ble::central::{impl#5}::to_raw @ /Users/haobogu/.cargo/git/checkouts/nrf-softdevice-03ef4aef10e777e4/d5f023b/nrf-softdevice/src/fmt.rs:151 
```

I checked the softdevice code, it seems the whitelist config is used in `scan_config.to_raw()`, in which `sd_ble_gap_whitelist_set` is called:

```rust
        // Set whitelist
        if let Some(w) = self.whitelist {
            assert!(w.len() <= u8::MAX as usize);
            let ret = unsafe { raw::sd_ble_gap_whitelist_set(w.as_ptr() as _, w.len() as u8) };
            if let Err(err) = RawError::convert(ret) {
                warn!("sd_ble_gap_whitelist_set err {:?}", err);
                return Err(err.into());
            }
            scan_params.set_filter_policy(raw::BLE_GAP_SCAN_FP_WHITELIST as _);
        } else {
            scan_params.set_filter_policy(raw::BLE_GAP_SCAN_FP_ACCEPT_ALL as _);
        }
```

However, `sd_ble_gap_whitelist_set` cannot be called when there's another connection.

From my perspective, `central::connect` should accepts an address, and use that address in `sd_ble_gap_connect`. Current implementation just fills `ptr::null()` to `sd_ble_gap_connect`:

```rust
async fn connect_inner<F>(_sd: &Softdevice, config: &ConnectConfig<'_>, new_conn: F) -> Result<Connection, ConnectError> {
    ...
    let ret = unsafe { raw::sd_ble_gap_connect(ptr::null(), &scan_params, &config.conn_params, 1) };  // <--- The whitelist is set in scan_params, and sd_ble_gap_connect takes ptr::null() as address
    if let Err(err) = RawError::convert(ret) {
        warn!("sd_ble_gap_connect err {:?}", err);
        return Err(err.into());
    }
}
```

I suggestion to have the following connect:

```rust
// Add addr as parameter
async fn connect_inner<F>(_sd: &Softdevice, addr: Option<[u8; 4]>, config: &ConnectConfig<'_>, new_conn: F) -> Result<Connection, ConnectError> { 
    ...
    // connect if addr is Some
    let ret = unsafe { raw::sd_ble_gap_connect(addr.unwrap_or(ptr::null), &scan_params, &config.conn_params, 1) }; 
    if let Err(err) = RawError::convert(ret) {
        warn!("sd_ble_gap_connect err {:?}", err);
        return Err(err.into());
    }
}

```

Dirbaio, 2024-11-15:
You can't do two connects concurrently, even with the fix you propose.

You have to do one connect with a whitelist with the two addrs. When you get one connection, check which addr it is, then start another connect with a whitelist with just the other addr.

HaoboGu, 2024-11-15:
okay, I'll try this approach

closed 2024-11-15 by HaoboGu