embassy-rs/nrf-softdevice #291: Link time error `duplicate symbol: __INTERRUPTS` on newest embassy-nrf and embassy-time versions

SRabia, 2025-02-10:
I run into an issue trying to build my project (**with microbit-v2**), It fail to link with the following error: 
```
ojects/embed/my-ble/target/thumbv7em-none-eabihf/debug/deps/my_ble-fc216197f6dc151f" "--gc-sections" "-Tlink.x" "-Tdefmt.x"
  = note: rust-lld: error: duplicate symbol: __INTERRUPTS
          >>> defined at lib.rs:88 (src/lib.rs:88)
          >>>            nrf52833_pac-f8fddcb5ae7af475.nrf52833_pac.16b0afa615586325-cgu.0.rcgu.o:(__INTERRUPTS) in archive /home/s/projects/embed/my-ble/target/thumbv7em-none-eabihf/debug/deps/libnrf52833_pac-f8fddcb5ae7af475.rlib
          >>> defined at pac.rs:145 (src/./chips/nrf52833/pac.rs:145)
          >>>            nrf_pac-ba52acb4edaae168.nrf_pac.299d165c7d8d431c-cgu.7.rcgu.o:(.vector_table.interrupts+0x0) in archive /home/s/projects/embed/my-ble/target/thumbv7em-none-eabihf/debug/deps/libnrf_pac-ba52acb4edaae168.rlib
```
my Cargo toml dependency: 
```toml

[dependencies]
cortex-m = {version = "0.7.7", features = ["critical-section-single-core"]}
cortex-m-rt = "0.7.5"
defmt = "0.3.10"
defmt-rtt = "0.4.1"
panic-probe = {version = "0.3.2", features = ["print-defmt"]}

# embassy
embassy-executor = {version = "0.7.0", features = ["arch-cortex-m", "defmt", "executor-thread", "executor-interrupt"]}
# embassy-nrf = {version = "0.3.1", features = ["defmt", "gpiote", "time-driver-rtc1", "time"]}
embassy-nrf = {version = "0.3.0", features = [
    "defmt", 
    "gpiote",
    "time", 
    "time-driver-rtc1", 
    "nrf52833",
]}

embassy-time = {version = "0.4.0", features = ["defmt", "defmt-timestamp-uptime"]}
embassy-futures = {version = "0.1.1", features = ["defmt"]}
embassy-sync = {version = "0.6.2", features = ["defmt"]}
lsm303agr = {version = "1.1.0", features = ["async"]}

# ble
nrf-softdevice = {version = "0.1.0", features = ["defmt", "s140", "ble-l2cap", "ble-sec",  "nrf52833", "ble-peripheral", "critical-section"]}

```
## my solution 
I didn't dig too much into the link error I have simply downgrade my `embassy-nrf`  to `0.2.0` and `embassy-time` to version `0.3.0` and things where building for me. 

I just want to leave this issue here for documentation and for other that have encounter the same issue as me.

SRabia, 2025-02-10:
as the [readme](https://github.com/embassy-rs/nrf-softdevice?tab=readme-ov-file#linking-issues) have to remove `critical-section` feature from cortex-m and add the feature `critical section-impl` to the softdevice crate.

alexmoon, 2025-02-10:
Embassy has moved to using nrf-pac instead of the nrf528xx-pac crates. Your dependencies are pulling in both PACs which is not supported.

nrf-softdevice does not depend on the PAC and will work with either version, so your problem has to do with some other crates. You can use `cargo tree` to examine which crates are pulling in which PACs.

SRabia, 2025-02-10:
Ah I see, I was using version "0.1.0"(nrf-softdevice) seem like it didn't remove the dependency to pac yet ( done in branch [no-pac,](https://github.com/embassy-rs/nrf-softdevice/tree/no-pac) I would guess?) So I update my dependency to point to master here is my final toml dependencies:
```toml

[dependencies]
# cortex-m = {version = "0.7.7", features = [""]}
cortex-m = "0.7.7"

cortex-m-rt = "0.7.5"
defmt = "0.3.10"
defmt-rtt = "0.4.1"
panic-probe = {version = "0.3.2", features = ["print-defmt"]}

# embassy
embassy-executor = {version = "0.7.0", features = ["arch-cortex-m", "defmt", "executor-thread", "executor-interrupt"]}
# embassy-nrf = {version = "0.3.1", features = ["defmt", "gpiote", "time-driver-rtc1", "time"]}
embassy-nrf = {version = "0.3.1", features = ["defmt", "gpiote","time", "time-driver-rtc1", "nrf52833", "unstable-pac"]}

embassy-time = {version = "0.4.0", features = ["defmt", "defmt-timestamp-uptime"]}
embassy-futures = {version = "0.1.1", features = ["defmt"]}
embassy-sync = {version = "0.6.2", features = ["defmt"]}
lsm303agr = {version = "1.1.0", features = ["async"]}

# ble

nrf-softdevice = { git = "https://github.com/embassy-rs/nrf-softdevice.git", branch = "master", features =[ "defmt", "s140", "ble-l2cap", "ble-sec",  "nrf52833", "ble-peripheral", "critical-section","critical-section-impl" ] }

```

I guess we can close the issue? I don't know  if others will make the same mistake as me? Because the latest version for nrf-softdevice is still "0.1.0"

closed 2025-02-11 by alexmoon

Wtoll, 2025-04-14:
Hi there, trying to use embassy for the first time and can confirm that others have made the same mistake. Glad I finally found the issue for this problem, but it would be really cool if this was better documented. It is kind of confusing having the latest version still just 0.1.0.