embassy-rs/nrf-softdevice #293: Unable to run on Adafruit Feather nRF52840 Sense

alistair23, 2025-02-28:
I'm trying to run an example on a [Adafruit Feather nRF52840 Sense](https://learn.adafruit.com/adafruit-feather-sense/overview).

I'm building an example using the following commands 

```shell
$ cd examples/
$ cargo run --bin ble_bas_peripheral --features nrf52840
...
     Running `probe-rs run --chip nRF52840_xxAA /var/mnt/scratch/alistair/software/tock/nrf-softdevice/target/thumbv7em-none-eabihf/debug/ble_bas_peripheral`
Error: No connected probes were found.
```

`probe-rs` is unable to flash the firmware as the sense only supports UF2, so I am generating a UF2 using

```shell
$ arm-none-eabi-objcopy -O ihex \
    target/thumbv7em-none-eabihf/debug/ble_bas_peripheral \
    target/thumbv7em-none-eabihf/debug/ble_bas_peripheral.hex
$ python3 ./uf2conv.py target/thumbv7em-none-eabihf/debug/ble_bas_peripheral.hex -c -f 0xADA52840
$ cp flash.uf2 /run/media/alistair/FTHRSNSBOOT/ ; sync
```
 
at which point the board resets, but there is no serial console exposed to the host and nothing seems to work.

I can get the board back into boot loader mode by double pressing the reset button and I can see that I have loaded the SoftDevice and the updated the boot loader. But the firmware never seems to run

```
UF2 Bootloader 0.9.2 lib/nrfx (v2.0.0) lib/tinyusb (0.12.0-145-g9775e7691) lib/uf2 (remotes/origin/configupdate-9-gadbb8c7)
Model: Adafruit Feather nRF52840 Sense
Board-ID: nRF52840-Feather-Sense
Date: Jul 19 2024
SoftDevice: S140 7.3.0
```

Any ideas on why this isn't working?

tmpfs, 2025-05-23:
Having just debugged this, you need to use `adafruit-nrfutil` to flash the device I think. I installed it using `uv` and then to package I used:

```bash
#!/usr/bin/env bash

set -e

error() {
	echo "$1" >&2
	exit 1
}

command -v cargo >/dev/null || error "cargo must be installed"
command -v adafruit-nrfutil >/dev/null || error "adafruit-nrfutil must be installed"

NAME="${1:-blinky}"
HEX="target/${NAME}.hex"
ZIP="target/${NAME}.zip"

echo "name: $NAME"
echo "hex: $HEX"
echo "zip: $ZIP"

rm -f target/$NAME.{hex,zip}

cargo objcopy --release --bin $NAME -- \
	-O ihex \
	"$HEX"

adafruit-nrfutil --verbose dfu genpkg \
	--dev-type 0x0052 \
	--sd-req 0x00B6 \
	--application "$HEX" \
	"$ZIP"
```

And then to flash:

```bash
#!/usr/bin/env bash

set -e

error() {
	echo "$1" >&2
	exit 1
}

command -v adafruit-nrfutil >/dev/null || error "adafruit-nrfutil must be installed"

NAME=${1:-blinky}
PKG="target/${NAME}.zip"
PORT=${2:-/dev/cu.usbmodem101}

echo "package: $PKG"
echo "port: $PORT"

adafruit-nrfutil --verbose dfu serial \
	--port "$PORT" \
	--baudrate 115200 \
	--package "$PKG" \
	--singlebank
```

In my case the board had Softdevice 6.1.1 so I had to offset by 152KB in `memory.x` but for you using the 7.3.0 you should configure 156KB in `memory.x` i believe, check the driver release notes to be certain.

tmpfs, 2025-05-23:
Note you will need to change the `--sd-req` argument to match your softdevice driver version!

alistair23, 2025-06-12:
Nothing seems to work.

I suspect it's just a dud board though, so I'm closing this

closed 2025-06-12 by alistair23